bertmelis/esp32ModbusRTU #1: interrupt watchdog timout on uploading example code

AkhileshThorat, 2019-02-18:
Hello,
I made few changes to the example application (changed function code, register address and serial pins)
On uploading I get the following message

Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400d0cdc: fcdc3191 3bb808a0 3bb80898
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9eac: 33313331 32313830 31333239
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9eac: 33313331 32313830 31333239
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9eac: 33313331 32313830 31333239
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9eac: 33313331 32313830 31333239
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9eac: 33313331 32313830 31333239
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9eac: 33313331 32313830 31333239
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9eac: 33313331 32313830 31333239
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9eac: 33313331 32313830 31333239
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9eac: 33313331 32313830 31333239
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9eac: 33313331 32313830 31333239
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9eac: 33313331 32313830 31333239
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9eac: 33313331 32313830 31333239
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9eac: 33313331 32313830 31333239
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9eac: 33313331 32313830 31333239
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9eac: 33313331 32313830 31333239
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9eac: 33313331 32313830 31333239
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9eac: 33313331 32313830 31333239
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9eac: 33313331 32313830 31333239
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9eac: 33313331 32313830 31333239
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9eac: 33313331 32313830 31333239
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9eac: 33313331 32313830 31333239
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9eac: 33313331 32313830 31333239
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9eac: 33313331 32313830 31333239
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9eac: 33313331 32313830 31333239
Guru Meditation Error: Core  1 panic'ed (IllegalInstruction). Exception was unhandled.
Memory dump at 0x400e9eac: 33313331 32313830 31333239
Guru Meditation ErrGuru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
Debug exception reason: Stack canary watchpoint triggered (loopTask) 
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Interrupt wdt timeout on CPU1)
Guru Meditation Error: Core  1 panic'ed (Guru Meditation Error: Core  1 panic'ed (Guru Meditation Error: Core  1 panic'ed (

bertmelis, 2019-02-18:
what do you mean with "on uploading"?
Can you share a (minimal) sketch?
What is your pin configuration?
Which board do you use?

AkhileshThorat, 2019-02-18:
I'm using ESP32 devkit V2

Pins used are 9,10(Rx,Tx) NOTE:(Worked on Micropython with this Pin configuration)

sketch is as follows:(made few changes inside your example application)

esp32ModbusRTU modbus(&Serial1, 33);  // use Serial1 and pin 16 as RTS

void setup() {
  Serial.begin(115200);  // Serial output
  Serial1.begin(19200, SERIAL_8N1, 9, 10, true);  // Modbus connection

  modbus.onData([](uint8_t serverAddress, MBFunctionCode fc, uint8_t* data, size_t length) {
    Serial.printf("id 0x%02x fc 0x%02x len %u: 0x", serverAddress, fc, length);
    for (size_t i = 0; i < length; ++i) {
      Serial.printf("%02x", data[i]);
    }
    std::reverse(data, data + 4);  // fix endianness
    Serial.printf("\nval: %.2f", *reinterpret_cast<float*>(data));
    Serial.print("\n\n");
  });
  modbus.onError([](MBError error) {
    Serial.printf("error: 0x%02x\n\n", static_cast<uint8_t>(error));
  });
  modbus.begin();

}

void loop() {
  static uint32_t lastMillis = 0;
  if (millis() - lastMillis > 30000) {
    lastMillis = millis();
    Serial.print("sending Modbus request...\n");
    modbus.readHoldingRegisters(0x01, 52, 2);
  }
}

bertmelis, 2019-02-18:
And the errors show up immidiately after uploading, or only after 30 seconds?

AkhileshThorat, 2019-02-18:
Hi,
I figured out the problem. It was causing due to QIO enabled flash(uses Pins 9,10 for flashing), which I later changed to DIO. Now I can use Pins 9,10 as Serial Pins for ModBus communication.Thank you for your valuable time.

bertmelis, 2019-02-18:
No problem! Glad you found the cause. (I'm short on time, finishing this lib is hard enough even without bugs 😟 )

AkhileshThorat, 2019-02-18:
Newer versions of Arduino and ESP32 cores comes with QIO enabled flash option as default, which i noticed after reading several issues.

AkhileshThorat, 2019-02-18:
> 
> 
> No problem! Glad you found the cause. (I'm short on time, finishing this lib is hard enough even without bugs 😟 )

My work is based majorly on Modbus communication( both Serial and TCP).I'm hugely benefited from your work.
When are you planning to interface LAN8720 with ESP32 for Modbus TCP?

closed 2019-02-18 by AkhileshThorat

bertmelis, 2019-02-18:
I'm struggling with ModbusTCP as my device doesn't connect reliably. It's coming, but I'm not giving any deadline 😉 
I want to connect my solar power inverter (with Modbus TCP) to log the yield.
No plans for LAN8720 as I don't have such a thing.

AkhileshThorat, 2019-02-18:
Sounds great.Even I'm interested in communicating with solar inverters using Modbus.
You should have a look at LAN8720,it's widely used in the community and supported by espressif systems.