bertmelis/esp32ModbusRTU #14: Getting error: 0xf7

gsaikiran29, 2020-03-04:
Hello,
Im new to esp32 i want to implement a modbus RTU master using esp32.
I took serial2 as modbus and made as per that. the esp32 initialized as master and it sending request to slave slave also responding but I am getting

sending Modbus request...<LF>
error: 0xf7<LF>

I cant find this f7 anywhere in the library files and from where it is printing. what is this error?

bertmelis, 2020-03-04:
Error 0xEF is indeed not defined. Could you share your code?

gsaikiran29, 2020-03-07:
```
/*

Copyright 2018 Bert Melis

This example reads 2 words (4 bytes) from address 52 of a server with id 1.
address 52 = register 30053 (Eastron SDM630 Total system power)
The ESP is connected to a max3485 with pins 17 (RX), 4 (TX) and 16 as RTS.

*/

#include <Arduino.h>
#include <esp32ModbusRTU.h>
#include <algorithm>  // for std::reverse

esp32ModbusRTU modbus(&Serial2);  // use Serial1 and pin 16 as RTS

void setup() {
  Serial.begin(115200);  // Serial output
  Serial2.begin(115200, SERIAL_8N1, 17, 16, true);  // Modbus connection

  modbus.onData([](uint8_t serverAddress, esp32Modbus::FunctionCode fc, uint16_t address, uint8_t* data, size_t length)
  {
    Serial.printf("id 0x%02x fc 0x%02x len %u: 0x", serverAddress, fc, length);
    for (size_t i = 0; i < length; ++i)
    {
      Serial.printf("%02x", data[i]);
    }
    std::reverse(data, data + 4);  // fix endianness
    Serial.printf("\nval: %.2f", *reinterpret_cast<float*>(data));
    Serial.print("\n\n");
  });
  modbus.onError([](esp32Modbus::Error error)
  {
    Serial.printf("error: 0x%02x\n\n", static_cast<uint8_t>(error));
  });
  modbus.begin();

}

void loop() {
  static uint32_t lastMillis = 0;
  if (millis() - lastMillis > 3000) {
    lastMillis = millis();
    Serial.print("sending Modbus request...\n");
    modbus.readInputRegisters(0x01, 0, 4);
  }
}
```

bertmelis, 2020-03-07:
The error comes from an error code returned by the modbus slave. See these lines:
https://github.com/bertmelis/esp32ModbusRTU/blob/f85417c00fb0d937a4a78a31750513c9bd84d3c8/src/ModbusMessage.cpp#L247

Could you switch lines A and B? (I know, my readme is not clear on this but so is the specification)

pcbtborges, 2020-07-19:
Hi, I am trying to read a RS485 distance meter and I am getting  error: 0xe0

I have used pins 16 and 17 for serial2 comm and pin 4 as
My code is as below:

```
/*

Copyright 2018 Bert Melis

This example reads 2 words (4 bytes) from address 52 of a server with id 1.
address 52 = register 30053 (Eastron SDM630 Total system power)
The ESP is connected to a max3485 with pins 17 (RX), 4 (TX) and 16 as RTS.

*/
#define MAX485_RE_NEG  4
#define RX_PIN        16 //RX2 
#define TX_PIN        17 //TX2 

#include <Arduino.h>
#include <esp32ModbusRTU.h>
#include <algorithm>  // for std::reverse

esp32ModbusRTU modbus(&Serial2, MAX485_RE_NEG);  // use Serial2 and pin 4 as RTS

void setup() {
  Serial.begin(115200);  // Serial output
  //Serial2.begin(9600, SERIAL_8N1, 17, 4, true);  // Modbus connection
  Serial2.begin(9600, SERIAL_8N1, RX_PIN, TX_PIN, true);

  modbus.onData([](uint8_t serverAddress, esp32Modbus::FunctionCode fc, uint16_t address, uint8_t* data, size_t length) {
    Serial.printf("id 0x%02x fc 0x%02x len %u: 0x", serverAddress, fc, length);
    for (size_t i = 0; i < length; ++i) {
      Serial.printf("%02x", data[i]);
    }
    std::reverse(data, data + 4);  // fix endianness
    Serial.printf("\nval: %.2f", *reinterpret_cast<float*>(data));
    Serial.print("\n\n");
  });
  modbus.onError([](esp32Modbus::Error error) {
    Serial.printf("error: 0x%02x\n\n", static_cast<uint8_t>(error));
  });
  modbus.begin();

}

void loop() {
  static uint32_t lastMillis = 0;
  if (millis() - lastMillis > 10000) {
    lastMillis = millis();
    Serial.print("sending Modbus request...\n");
    modbus.readInputRegisters(0x01, 00, 2);             //changed to address 00 as per attached manual
  }
}
```
For your reference please check device included device manual.

[mhmodbus-rtu （Ultrasonic level sensor RS485 ）.pdf](https://github.com/bertmelis/esp32ModbusRTU/files/4943062/mhmodbus-rtu.Ultrasonic.level.sensor.RS485.pdf)
[image]

Assistance highly welcome.
Thanks in advance
Paulo

pcbtborges, 2020-07-19:
Please find the device manual below:
[mhmodbus-rtu （Ultrasonic level sensor RS485 ）.pdf](https://github.com/bertmelis/esp32ModbusRTU/files/4943072/mhmodbus-rtu.Ultrasonic.level.sensor.RS485.pdf)

bertmelis, 2020-07-22:
How does your circuit look like? What parts do you use?

I tested with
- a (genuine) MAX3485
- 120 Ohm resistor on both ends of the RS485 wire
- 680 Ohm biasing resistors on  A and B lines (B to 3.3V and A to GND)
- Connect GND of the ESP via a 100 Ohm resistor to the GND of the device

Error 0E indicates a timeout. Your sensor isn't receiving the data or isn't responding. You could attach a logic analyzer to the RX/TX of the ESP (NOT the RS485 line!) and see what is going out/coming in.

pcbtborges, 2020-07-22:
Dear bertmelis, thanks a for taking the time to respond to my message.

[image]

That was the basis for my circuit, just attach the distance meter in place of this power meter on the picture.
ESP32 + MAX485 boards.

I have done some tests.
Installed a 250ohms resistor between ground and 4-20ma line and attached a voltmeter in parallel.
Voltage varies when I point the distance meter to walls at different distances so I assume the distance meter is working.

Also changed the code to send some commands on the line: modbus.readInputRegisters(0x01, 00, 2);
(like 0x00) and the result indicated invalid command. 

I also have ordered an USB to RS485 converter to try to see what is actually coming from the Distance meter and it is to arrive soon.

Thanks again
Paulo

bertmelis, 2020-07-22:
Can you swap A and B? on the MAX485 board?

If I found the right circuit on the Internet, this board biases the A line to Vcc and the B line to ground (can be checked with your multimeter) whereas it should be opposite.

bertmelis, 2020-07-22:
See also http://e2e.ti.com/cfs-file/__key/telligent-evolution-components-attachments/13-143-00-00-00-26-49-60/RS485-_2D00_-Polarity-Conventions.pdf

Mind that your ESP uses UART and treats HIGH as binary 1 and LOW as binary 0. That's why the Serial needs to be "inverted".

*If somebody reads along, please correct me if I'm wrong. It's been a long time since I left school*

pcbtborges, 2020-07-22:
Hi, did invert A/B and there is a change in another code I have that uses ModbusMaster.h (//https://github.com/4-20ma/ModbusMaster) instead of (esp32ModbusRTU.h)

Originally the former returned "illegal Function" but when I change A<->B then it starts returning "Response timed out"

Using esp32ModbusRTU.h with A and B as originally described:
sending Modbus request...
error: 0xe0

Swapping A<->B:
No change.
I did make a change on the the code that might be affecting the sketch:
```
  //Serial2.begin(9600, SERIAL_8N1, 17, 4, true);  // Modbus connection
  Serial2.begin(9600, SERIAL_8N1, RX_PIN, TX_PIN, true);  (See sketch above)
```
I did not figure out why the 4 on the line above as I am connecting with Serial2 pins 16 and 17.
I did add:
```
  pinMode(4, OUTPUT);
  // Init in receive mode
  digitalWrite(4, HIGH);  (tried high and low but no change)
```

As I said, I ordered a USB to RS485 adapter to check communication between the PC and the Distance Meter and it is to arrive soon.

Thanks 
Paulo

bertmelis, 2020-07-22:
Make sure you have this

```C++
#define MAX485_RE_NEG  4
#define Slave_ID       1
#define RX_PIN        16
#define TX_PIN        17

esp32ModbusRTU modbus(&Serial1, MAX485_RE_NEG);

// in void setup()
Serial1.begin(9600, SERIAL_8N1, RX_PIN, TX_PIN, true);
```

pcbtborges, 2020-07-22:
Hi, DE & RE as in the Fritzing diagram above.

Code as of now:

```
/*
Copyright 2018 Bert Melis

This example reads 2 words (4 bytes) from address 52 of a server with id 1.
address 52 = register 30053 (Eastron SDM630 Total system power)
The ESP is connected to a max3485 with pins 17 (RX), 4 (TX) and 16 as RTS.
*/

#define RX_PIN        16 //RX2 
#define TX_PIN        17 //TX2 
#define MAX485_RE_NEG  4
#define Slave_ID       1

#include <Arduino.h>
#include <esp32ModbusRTU.h>
#include <algorithm>  // for std::reverse

esp32ModbusRTU modbus(&Serial2, MAX485_RE_NEG);

void setup() {
  Serial.begin(115200);
  Serial2.begin(9600, SERIAL_8N1, RX_PIN, TX_PIN, true);  // Modbus connection

  modbus.onData([](uint8_t serverAddress, esp32Modbus::FunctionCode fc, uint16_t address, uint8_t* data, size_t length) {
    Serial.printf("id 0x%02x fc 0x%02x len %u: 0x", serverAddress, fc, length);
    for (size_t i = 0; i < length; ++i) {
      Serial.printf("%02x", data[i]);
    }
    std::reverse(data, data + 4);  // fix endianness
    Serial.printf("\nval: %.2f", *reinterpret_cast<float*>(data));
    Serial.print("\n\n");
  });
  modbus.onError([](esp32Modbus::Error error) {
    Serial.printf("error: 0x%02x\n\n", static_cast<uint8_t>(error));
  });
  modbus.begin();

}

void loop() {
  static uint32_t lastMillis = 0;
  if (millis() - lastMillis > 10000) {
    lastMillis = millis();
    Serial.print("sending Modbus request...\n");
    modbus.readInputRegisters(0x00, 02, 2);
  }
}
```

Console output still:
sending Modbus request...
error: 0xe0

USB to RS485 will arrive tomorow and will provide additional info to aid finding the problem.

bertmelis, 2020-07-22:
What you can do is use `Serial` as Modbus interface, connect the board via USB to your computer and look in your terminal what the ESP is actually sending. The ESP should be sending a modbus message. Mind that you don't `invert` the signal.

I use Termite (in Windows) for that, with the Hexadecimal view plugin (https://www.compuphase.com/software_termite.htm). You might as well use Realterm.

pcbtborges, 2020-07-22:
Do you mean the A and B from the USB to Serial adapter in parallel with the Arduino Serial2 (in my case)?
Sounds like a good idea.
I have Putty in my PC, I will try it and see if it works otherwise go to one of your suggestions.
Thanks

bertmelis, 2020-07-23:
I'm not completely following...

When you pass Serial instead of Serial1 to the modbus object, the modbus messages will show up on your terminal. You'll want to display them in hex instead of ASCII, but that depends on your terminal software.

You then just leave out Serial1 as it is not used.

pcbtborges, 2020-07-23:
I am also a bit lost here.
ESP32 has 3 serial. I am using "Serial", that is connected to the USB and the PC as serial console monitor.
I am using "Serial2" on GPIO16 and 17 to connect to the RS232 to RS485 (*) converter as shown on the sketch above.
I am not using "Serial1" in my project.

I understood your suggestion to be attaching  the USB to RS485 converter that is arriving on the PC and the A&B wires to the A&B wired of the Distance meter (that are currently connected to the RS232 to RS486 (*) converter to see what the ESP32 is sending and receiving as the sketch executes.
Is that what you meant?

In any case I will first connect the USB to RS485 adapter to the PC and A&B to the Distance meter and send the commands directly to it and see if it is working as expected.

Then I will try the connection above or even remove the distance meter all together and try the two RS485 between them to see what happens.

Will keep you posted, thanks so far
Paulo

pcbtborges, 2020-07-24:
I have got the RS485 to USB converter but no good comunication so far.
My circuit:

[image]

[image]

First I tested with a simple sketch to transfer console input:
```
#define TX             17
#define RX             16

void setup() {
  Serial.begin(115200);
  Serial2.begin(9600, SERIAL_8N1, RX, TX);
}

void loop() {
  char text[1];
  if (Serial2.available()) {
    Serial.write(Serial2.read());
  }
  if (Serial.available()) {
    char c = Serial.read();
    Serial2.write(c);
    Serial.write(c);
  }
}
```
Output as below:
[image]

What I type on the Serial Console goes through the ESP32 and reaches Termite.
What I type on Termite is not shown on the ESP32 console.

Now with the main sketch as below:
```
/*
Copyright 2018 Bert Melis

This example reads 2 words (4 bytes) from address 52 of a server with id 1.
address 52 = register 30053 (Eastron SDM630 Total system power)
The ESP is connected to a max3485 with pins 17 (RX), 4 (TX) and 16 as RTS.
*/

#define RX_PIN        16 //RX2 
#define TX_PIN        17 //TX2 
#define Slave_ID       1
#define MAX485_RE_NEG  4

#include <Arduino.h>
#include <esp32ModbusRTU.h>
#include <algorithm>  // for std::reverse

esp32ModbusRTU modbus(&Serial2, MAX485_RE_NEG);

void setup() {
  Serial.begin(115200);
  Serial2.begin(9600, SERIAL_8N1, RX_PIN, TX_PIN, true);  // Modbus connection

  modbus.onData([](uint8_t serverAddress, esp32Modbus::FunctionCode fc, uint16_t address, uint8_t* data, size_t length) {
    Serial.printf("id 0x%02x fc 0x%02x len %u: 0x", serverAddress, fc, length);
    for (size_t i = 0; i < length; ++i) {
      Serial.printf("%02x", data[i]);
    }
    std::reverse(data, data + 4);  // fix endianness
    Serial.printf("\nval: %.2f", *reinterpret_cast<float*>(data));
    Serial.print("\n\n");
  });
  modbus.onError([](esp32Modbus::Error error) {
    Serial.printf("error: 0x%02x\n\n", static_cast<uint8_t>(error));
  });
  modbus.begin();

}

void loop() {
  static uint32_t lastMillis = 0;
  if (millis() - lastMillis > 10000) {
    lastMillis = millis();
    Serial.print("sending Modbus request...\n");
    modbus.readInputRegisters(0x00, 02, 2);
  }
}
```
[image]

Did not find a reason for, in the first test with just a serial input and Termite input, why what I type on Termite is not echoed on the ESP32 serial console.

Also cannot understand what the main sketch is sending to the Termite terminal.

Can you make something out of it?
Thanks
Paulo

bertmelis, 2020-07-24:
You've connected the level converter wrong. high voltage is MAX485 side, low is ESP. The MAX485 should be powered with 5V, not 3.3V. You might as well fried your pins doing this.

pcbtborges, 2020-07-24:
Sorry, my mistake on the drawing, level converter is inverted on the drawing.
MAX485 is been powered by 5v.
Here is the corrected diagram:

[image]

Paulo

pcbtborges, 2020-07-25:
Still testing with the manual input of commands.
With the following code I can, but not simultaneously, send data in both directions for testing purposes:

```
#define MAX485_RE_NEG  4
#define TX             17
#define RX             16

void setup() {
  Serial.begin(115200);
  Serial2.begin(9600, SERIAL_8N1, RX, TX);
  pinMode(MAX485_RE_NEG, OUTPUT);
  // Init in receive mode
  //digitalWrite(MAX485_RE_NEG, LOW);     //When enabled Termite to Serial Console OK
  digitalWrite(MAX485_RE_NEG, HIGH);      //When enabled Serial Console to Termite OK,
                                                                    //To work as defined above only one can be active at a time
}

void loop() {
  if (Serial2.available()) {   
    while (Serial2.available()){ 
      Serial.write(Serial2.read());
    } 
  }
  if (Serial.available()) {
    while(Serial.available()){
      char c = Serial.read();
      Serial2.write(c);
      Serial.write(c);
    }  
  }
}
```

bertmelis, 2020-07-25:
Strange. Maybe some task priority issues. I'll check things out. You can fiddle yourself if you alter the task priority directly in the code.

bertmelis, 2020-07-25:
Just to be sure: how much time passes between sending the request and receiving the error? Should be about 5 secs.

pcbtborges, 2020-07-25:
I am not that advanced, don't know how to change task priority.
It does not feel it takes that long, looks like it is immediate.

pcbtborges, 2020-07-25:
Just tested again.
From the moment I see the message "sending Modbus request..." on the console screen to see the "ff f7 ff fb ff fb 5d 89                          ÿ÷ÿûÿû]‰   " at Termite takes less than 1 second. From te Termite message showing to the Console screen showing "error: 0xe0" about 5 seconds as you anticipated.

pcbtborges, 2020-07-25:
Just tested with another sketch I have for modbus.
Code as below:

```
#include "ModbusMaster.h" //https://github.com/4-20ma/ModbusMaster
/*!
  http://www.bizkit.ru/en/2019/02/21/12563/
  We're using a MAX485-compatible RS485 Transceiver.
  Rx/Tx is hooked up to the hardware serial port at 'Serial'.
  The Data Enable (DE) and Receiver Enable (RE) pins are hooked up as follows:
*/
#define MAX485_RE_NEG  4 //D4 RS485 has a enable/disable pin to transmit or receive data. Arduino Digital Pin 2 = Rx/Tx 'Enable'; High to Transmit, Low to Receive
#define Slave_ID       1
#define RX_PIN        16 //RX2 
#define TX_PIN        17 //TX2 
// instantiate ModbusMaster object
ModbusMaster modbus;
void preTransmission(){
  digitalWrite(MAX485_RE_NEG, HIGH); //Switch to transmit data
}
void postTransmission(){
  digitalWrite(MAX485_RE_NEG, LOW); //Switch to receive data
}
long lastMillis = 0;

void setup(){
  pinMode(MAX485_RE_NEG, OUTPUT);
  // Init in receive mode
  digitalWrite(MAX485_RE_NEG, LOW);
  Serial.begin(115200);
  // Modbus communication runs at 9600 baud
  Serial2.begin(9600, SERIAL_8N1, RX_PIN, TX_PIN);
  modbus.begin(Slave_ID, Serial2);
  // Callbacks allow us to configure the RS485 transceiver correctly
  modbus.preTransmission(preTransmission);
  modbus.postTransmission(postTransmission);
}

void loop() {
  long currentMillis = millis();
  if (currentMillis - lastMillis > 10000)   {
    uint8_t result = modbus.readInputRegisters(0x00, 1);

    Serial.print("Result: ");
    Serial.println(result);
    
    if (getResultMsg(&modbus, result))    {
      Serial.println();
      double res_dbl = modbus.getResponseBuffer(0);
      String res = "Distance: " + String(res_dbl) + " m\r\n";
      Serial.println(res);
    }
    lastMillis = currentMillis;
  }
}

bool getResultMsg(ModbusMaster *node, uint8_t result) {
  String tmpstr2 = "\r\n";
  switch (result){
  case node->ku8MBSuccess:
    return true;
    break;
  case node->ku8MBIllegalFunction:
    tmpstr2 += "Illegal Function";
    break;
  case node->ku8MBIllegalDataAddress:
    tmpstr2 += "Illegal Data Address";
    break;
  case node->ku8MBIllegalDataValue:
    tmpstr2 += "Illegal Data Value";
    break;
  case node->ku8MBSlaveDeviceFailure:
    tmpstr2 += "Slave Device Failure";
    break;
  case node->ku8MBInvalidSlaveID:
    tmpstr2 += "Invalid Slave ID";
    break;
  case node->ku8MBInvalidFunction:
    tmpstr2 += "Invalid Function";
    break;
  case node->ku8MBResponseTimedOut:
    tmpstr2 += "Response Timed Out";
    break;
  case node->ku8MBInvalidCRC:
    tmpstr2 += "Invalid CRC";
    break;
  default:
    tmpstr2 += "Unknown error: " + String(result);
    break;
  }
  Serial.println(tmpstr2);
  return false;
}
```

Console and Termite as below:

[image]

In this case, when the distance meter is connected to A&B instead of the USB to RS485 addapter the error is: "Illegal Function"

Thanks
Paulo

bertmelis, 2020-07-25:
I might spotted a but in the Arduino core. I'll try this evening.

bertmelis, 2020-07-25:
Can you try after changing the false to true on this line:

https://github.com/espressif/arduino-esp32/blob/4d98cea085d619bed7026b37071bd8402a485d95/cores/esp32/esp32-hal-uart.c#L362

pcbtborges, 2020-07-25:
My esp32-hal-uart.c looks different:

```
void uartFlush(uart_t* uart)
{
    if(uart == NULL) {
        return;
    }

    UART_MUTEX_LOCK();
    while(uart->dev->status.txfifo_cnt || uart->dev->status.st_utx_out);

    //Due to hardware issue, we can not use fifo_rst to reset uart fifo.
    //See description about UART_TXFIFO_RST and UART_RXFIFO_RST in <<esp32_technical_reference_manual>> v2.6 or later.

    // we read the data out and make `fifo_len == 0 && rd_addr == wr_addr`.
    while(uart->dev->status.rxfifo_cnt != 0 || (uart->dev->mem_rx_status.wr_addr != uart->dev->mem_rx_status.rd_addr)) {
        READ_PERI_REG(UART_FIFO_REG(uart->num));
    }

    xQueueReset(uart->queue);

    UART_MUTEX_UNLOCK();
}
```

bertmelis, 2020-07-25:
Arduino? Platformio? Something else? Which version?

pcbtborges, 2020-07-26:
Arduino IDE

pcbtborges, 2020-07-26:
Hi, some more tests that may be of interest:
[image]
[image]
[image]
[image]

As you see with the USB to RS485 converter and using this test software I found it is possible to send and receive messages from the ultrasonic distance meter but curiously not by sending 0x04 as specified on the manual as to read input register.

I am trying to consult the manufacturer on that but at least indicate the device seems to be working ok.

bertmelis, 2020-07-26:
Thank you for testing. I can't find what's wrong though. The following weeks I'll be on holiday so I have very limited access. But I'll try to support.

pcbtborges, 2020-07-26:
Hi, thanks for your time. If you get something new it will be of help. 
I will keep trying to get something as I need to have this thing operational one way or the other. 
If I happen to find the solution I will post it here. 
Have a good holiday.