bertmelis/esp32ModbusRTU #15: Allow to not use any RTS pin

greg-lielens, 2020-03-29:
Hi, I just downloaded your lib and I managed to use it (goal is to read modbus electric and water meters).

Things are working nicely, but I have a XY-K485 TTL module to convert serial to RS485.
This do not use RTS pin, on RX and TX pins, I think it does it internally by buffering the serial RX/TX data and doing the RTS magic himself.
Anyway, I can send  order and get back the response from my water meter, using a non-connected RTS pin.
I do not like to loose pin though, even if it would not be a problem for my current project, so I changed your code so instead of exiting, providing a negative pin number simply skip all RTS pin operation (pinmode and write to high or low). Still working without any issue, so I though this cound be added to your lib....I can send a patch (although the mods needed are really trivial).
BTW first I though of using a non-existing pin instead of -1, like 100. It worked too, but it's not a good idea apparently, no guarantee it does not write to arbitrary memory address...

bertmelis, 2020-04-06:
Why would using `-1` not work? digitalWrite takes an unsigned int and by using -1 it transfers into 127 which is ignored. Same goes for 100.

See here:
https://github.com/espressif/arduino-esp32/blob/0f772270fbb32099cad72053755d6b4c7e8e1f82/cores/esp32/esp32-hal-gpio.h#L70
and here:
https://github.com/espressif/arduino-esp32/blob/0f772270fbb32099cad72053755d6b4c7e8e1f82/cores/esp32/esp32-hal-gpio.c#L176

greg-lielens, 2020-04-07:
There is an explicit abort in your code if pin<0...That's what I changed
(do all write to pin or set pin mode only if pin>=0).
It seems, that using a non-existing pin is dangerous on Arduino in general,
as it can write to non-pin-mapped memory thushave random consequences:
https://forum.arduino.cc/index.php?topic=43936.0
But indeed, looking at the code you sent, it seems the ESP32 prevent that,
it does not set mode or write to invalid pins....
My gut feeling is that explicitely skipping negative pin is clearer and
maybe safer in the long run (in the very unlilkely case that the check got
removed in newer ESP32-arduino updates)

BTW, I saw that you use SDM630 too.
I am trying to communicate to this one in addition to my modbus water
meter, on the same RS485 line.
I get some messages back, but garbled and all is quite unstable it seems,
even if I connect only the SDM630.
Did you have to terminate the A+/B-/Gnd line with resistors? And I saw you
invert the serial. I did not and it does not work if I do...Is it
equivalent to permutting A+/B-, or is there another reason you did that? I
also try to get multiple consecitive register simultaneously to reduce the
number of message exchanged....Did you try that?
Any additional info (.ino, connection drawing,...) you have on ESP32-SDM630
connection using your lib would be of great help ;-)

bertmelis, 2020-04-07:
Oh :flushed: I didn't know that I had "forbidden" that. Will fix today!

I indeed use 120Ohm resistors (on both ends) and connected ground via a resistor as in the schematic. Be sure to only place them on the line endings.
From what I remember, the A and B markings are swapped. See https://github.com/bertmelis/esp32ModbusRTU/issues/6

greg-lielens, 2020-04-07:
> Oh 😳 I didn't know that I had "forbidden" that. Will fix today!

Thanks a lot, it should help :-)

closed 2020-05-05 by bertmelis