bertmelis/esp32ModbusRTU #16: cast value

buddhafragt, 2020-04-04:
Hello,
I try to get a PZEM-003 working, the data value I receive is 2 x byte. I try it with this code:

`  modbus.readInputRegisters(0x02, 0x0000, 1); `

```
modbus.onData([](uint8_t serverAddress, esp32Modbus::FunctionCode fc, uint16_t address, uint8_t* data, size_t length) {
 ValueRead = 0;
  Serial.printf("received data: id: %u, fc %u\ndata: 0x", serverAddress, fc);
for (size_t i = 0; i < length; ++i) {
  Serial.printf("%02x", data[i]);
}
Serial.print("\n");
ValueRead = *reinterpret_cast<uint8_t*>(data);
  });

```
The data value I recive is "0x04cf" and this is right. 
But with "reinterpret_cast" it is not working. Someone have a hint for me?

Greetings, Michael

bertmelis, 2020-04-04:
You cast it to an uint8_t pointer. So dereferencing that returns the 8 bit value.

What type is ValueRead?

buddhafragt, 2020-04-04:
Oh, how stupid, I change to uint16_t and it is working, thank you!

I have another problem now: I have 2 devices on the Modbus, 1 with SERIAL_8N2 and other with SERIAL_8N1
So I try to change the serial parameter on the fly in the code. But no sucess. I try to change like this:
 ```
Serial1.flush();
  Serial1.begin(9600, SERIAL_8N2, 17, 4, false);  // Modbus connection: SERIAL_8N2 für PZEM-003
     delay(1000);
  modbus.readInputRegisters(0x02, 0x0000, 1); 
```
But now no values coming in again.....

bertmelis, 2020-04-04:
Can you confirm it works if you connect the separately?

buddhafragt, 2020-04-04:
Yes, booth work with different serial parameters: SERIAL_8N2 and SERIAL_8N1.

bertmelis, 2020-04-06:
There's nothing this lib can do for your problem but I would do some experiments:
-try both separately on different buses
- try both separately but on same bus (don't mix in one sketch)
- head over to Arduino-esp32 repo and try to resolve there
- Attach a logic analyzer and see if signals are coming through (beware of voltages!).

Are you sure you can't change you device's settings?

bertmelis, 2020-04-07:
That should be possible.

bertmelis, 2020-04-07:
An idea:

Can you just make the firmware print messages via Serial to your computer with 2 different baudrates. And monitor on your PC once with the first baudrate and again with the second. You should see first garbage + text and second text + garbage.

buddhafragt, 2020-04-10:
switching between 2 different baudrates in Loop dont work!

bertmelis, 2020-04-11:
Do you have the latest version of the Arduino core? Do you use Arduino IDE or something else (like VSCode + Platformio)?

I'm happy to support if it doesn't take too much of my time. But there's no apparent issue with this code. You might have better luck creating an issue in the Arduino for esp32 repo.

closed 2020-04-11 by bertmelis

buddhafragt, 2020-04-11:
Thanks for your time,
I solve the problem with 2 different ports for the devices and 2 instances.