bertmelis/esp32ModbusRTU #20: Rubbish instead of proper data

Miq1, 2020-05-17:
Hi,

I am trying to communicate through a MAX485 module with a power meter. I am using UART2 to drive it and your library of course.

The relevant code snippets are:

```
...
// RS485 MODBUS RTU
#include <esp32ModbusRTU.h>
esp32ModbusRTU RS485(&Serial2, GPIO_NUM_15);
...
setup() {
...
Serial2.begin(19200, SERIAL_8N1, GPIO_NUM_16, GPIO_NUM_17, true);
...
// Init RS485 MODBUS RTU
  RS485.onData(handleData);
  RS485.onError([](esp32Modbus::Error error) {
    Serial.printf("RS485 error: 0x%02x\n\n", static_cast<uint8_t>(error));
    });
  RS485.begin();

  // Test
  RS485.readHoldingRegisters(0x01, 0x0734, 4);
}
...
```
All I get is strange data on the Tx and no response from the meter. The meter's comms LED flashes shortly, though, when the request is transmitted.

With a logic analyzer I seem to get these data: 
```
52795-52796 UART: TX bits: 0
52796-52797 UART: TX bits: 0
52797-52798 UART: TX bits: 0
52798-52799 UART: TX bits: 0
52799-52800 UART: TX bits: 0
52800-52801 UART: TX bits: 1
52801-52802 UART: TX bits: 0
52802-52803 UART: TX bits: 1
52806-52807 UART: TX bits: 0
52807-52808 UART: TX bits: 0
52808-52809 UART: TX bits: 0
52809-52810 UART: TX bits: 0
52810-52811 UART: TX bits: 0
52811-52812 UART: TX bits: 1
52812-52813 UART: TX bits: 0
52813-52814 UART: TX bits: 1
52818-52819 UART: TX bits: 0
52819-52820 UART: TX bits: 0
52820-52821 UART: TX bits: 0
52821-52822 UART: TX bits: 1
52822-52823 UART: TX bits: 0
52823-52824 UART: TX bits: 0
52824-52825 UART: TX bits: 0
52825-52826 UART: TX bits: 1
52831-52832 UART: TX bits: 0
52832-52833 UART: TX bits: 1
52833-52834 UART: TX bits: 0
52834-52835 UART: TX bits: 0
52835-52836 UART: TX bits: 0
52836-52837 UART: TX bits: 0
52837-52838 UART: TX bits: 0
52838-52839 UART: TX bits: 0
52845-52846 UART: TX bits: 0
52846-52847 UART: TX bits: 1
52847-52848 UART: TX bits: 0
52848-52849 UART: TX bits: 0
52849-52850 UART: TX bits: 0
52850-52851 UART: TX bits: 0
52851-52852 UART: TX bits: 0
52852-52853 UART: TX bits: 1
52860-52861 UART: TX bits: 0
52861-52862 UART: TX bits: 0
52862-52863 UART: TX bits: 0
52863-52864 UART: TX bits: 1
52864-52865 UART: TX bits: 0
52865-52866 UART: TX bits: 1
52866-52867 UART: TX bits: 1
52867-52868 UART: TX bits: 0
```
There is no similarity to the intended data, I am afraid...

Do you have any suggestion as to where my error may be?

bertmelis, 2020-05-17:
Tried to swap A and B?

Miq1, 2020-05-17:
Yes, I did. No significant difference to be observed.

Is there an easy way to dump out the request packets from your lib? I found only private functions where I could do that and did not want to interfere with the task timings yet.

bertmelis, 2020-05-17:
You can add following lines here: https://github.com/bertmelis/esp32ModbusRTU/blob/72dd86c0a0fa020c73ca4bf4b73653997a280f20/src/esp32ModbusRTU.cpp#L111

```C++
for (size_t i = 0; i < length; ++i) Serial.printf("%02x ", data[i);
Serial.print("\n");
```

You'll need to include Arduino.h somewhere, otherwise it won't compile. Obviously, it's only in case if you read via `Serial`.

Miq1, 2020-05-17:
Thanks! I shied the edit, but that would have been the spot I had put in the dump.
I am off my workshop now but will try tomorrow.
I had the logic analyzer on the RO, DI and DE/RE pins, trigger was on flanks on the DE/RE pin. I will give the relevant schematics tomorrow as well.

bertmelis, 2020-05-17:
You could just hook up the logic analyser on the UART lines (RX/TX) to rule out any hardware problems. I have had bad luck with MAX485 until I found genuine MAX3485 chips.

Just to let you know: I don't have any modbus RTU devices myself at the moment so I haven't used the library for quite a while now. Maybe something changed in Serial/UART in the Arduino core...

Miq1, 2020-05-18:
The packet dump looks good:
```
1 3 7 34 0 4 4 B3 
RS485 error: 0xe0
```
Today the logic analyzer is in a better mood as well for whatever reason:
[image]
 
So I will have to look further down the line. A fellow user in another ESP32 forum mentioned it could be the MAX485 module spoiling the data, as it is specified for 5V, not for the 3.3V I am supplying it. I have no idea why that should happen, but to be on the safe side I ordered real 3.3V modules.

Miq1, 2020-05-18:
Oh, by the way: the ordered modules seem to have "auto DE" - I suppose the declaration of the esp32ModbusRTU object would be `esp32ModbusRTU RS485(&Serial2, -1);` then?

bertmelis, 2020-05-18:
> 
> 
> Oh, by the way: the ordered modules seem to have "auto DE" - I suppose the declaration of the esp32ModbusRTU object would be `esp32ModbusRTU RS485(&Serial2, -1);` then?

yes.

FYI: I first had MAX3485 modules from a big Chinese online selling platform. I guess I actually bought MAX485 modules because they didn't work. After replacing them by modules from a trusted seller, problems were gone.

Miq1, 2020-05-19:
I am progressing. The mentioned user suggested to use another MAX485 to listen on the bus and have the LA pick up the data from the RO line. That worked indeed and I found the power meter is answering, (LSB first):
[image]

And wonderously my ESP now is getting data as well:
```
1 3 7 34 0 4 4 B3 
received data: id: 1, fc 3
data: 0x001405730a1203e8
```

I'd swear any oath you like I did not change anything else but to attach the second module to the bus! And as soon as I plug the power of the second module, the old error pops up again. 
Well, the solution seems to be bad wiring - I mistook the D0/D1 connectors on the power meter to be A+/B-, respectively. In fact it seems to be opposite, D0 is B- and D1 is A+. My pullup/pulldown resistors at the first module were attached inversely as well, I was pulling down A+ and pulling up B-. So no communication could be stablished unless the second module with crossed A/B lines changed the levels again to be interpreted.

I hope I got it right now. Thanks you so much for your help!

closed 2020-05-19 by Miq1

bertmelis, 2020-05-19:
I should clarify the line names and the 'invert' in the readme...

But then again, not every vendor labels the connections the same.

Anyway, I'm glad it worked out for you.