bertmelis/esp32ModbusRTU #23: Getting error: 0xe0

Tsaadon, 2020-06-05:
Hey,
I used the example you provided, changed only the address and slave ID but I get this error, what does it mean?
Thanks.

bertmelis, 2020-06-05:
Timeout. Try to swap A and B wires.

Tsaadon, 2020-06-05:
Tried, still the same error :\, I know for sure the sensor is good because I connected it to the PC using a USB adapter and Modbus tester software and I get good output. Do you have any idea what else could be causing it?
According to the producer (and what worked using the PC), Slave ID 1,  I need to read address 0x0000-0x0002, Buadrate 9600, 8 data bits, 1 stop, no parity.

bertmelis, 2020-06-05:
Well, if you're getting a timeout, the esp is not receiving any data.

Are you using a line driver (max3485 or similar)? Which kind? Do you have a logic analyzer?

Tsaadon, 2020-06-06:
Yes, I'm using the max3485 by SparkFun, and no, no logic analyzer on hand.
https://www.sparkfun.com/products/10124
When the "sending Modbus request.." appears on the serial monitor the RTS LED on the max3485 blinks, if that helps in anyway..

Miq1, 2020-06-11:
Did you connect A and B only or GND as well? Did you terminate the RS485 on the max3485 module side (120 Ohms between A and B, 1kOhms between A<->Vcc and B<->GND)? Both may stabilize the signal. If you have a multimeter at hand, try to measure the voltage difference between A and B. It should vary when you are sending a request as well if a response is received.

bertmelis, 2020-06-11:
I did like in the readme, so yes I used the termination resistors as well as biasing resistors. I also connected GND. Keep in mind that I've not drawn the "other side of the line". There must also be a 120Ohm resistor.

I didn't use a breakout board but a bare max3485. Maybe your board already has some resistors.

Miq1, 2020-06-11:
:) 
I was referring to the post of Tsaadon, not yours. Sorry, should have made that clearer.

I personally have added the resistors despite the fact that I used a module - it resulted in a clearer signal.

bertmelis, 2020-06-11:
Oops, didn't notice. Working on phone while enjoying a view on the Tatra mountains.

Tsaadon, 2020-06-11:
I didn't connect GND and tried adding the resistors, we tried the voltmeter, and played a bit with turning from send to receive and it seems to work, no idea what the issue is. Some friends helped me with a different code (but the same library) and now it's working. Thanks for the help guys!

closed 2020-06-12 by bertmelis