bertmelis/esp32ModbusRTU #24: get Slave address and etc inside Error handler?

zekageri, 2020-06-15:
Hi! I just want to know if it is possible to get the address of the slave device wich is sending the error.

like:

`modbus.onError([](esp32Modbus::Error error, uint8_t serverAddress) {});`

Thanks.

zekageri, 2020-06-15:
### Okay! I was able to compile it like this:

**_In esp32ModbusRTU.cpp:_**

```
void esp32ModbusRTU::_handleConnection(esp32ModbusRTU* instance) {
  while (1) {
    ModbusRequest* request;
    xQueueReceive(instance->_queue, &request, portMAX_DELAY);  // block and wait for queued item
    instance->_send(request->getMessage(), request->getSize());
    ModbusResponse* response = instance->_receive(request);
    if (response->isSucces()) {
      if (instance->_onData) instance->_onData(response->getSlaveAddress(), response->getFunctionCode(), request->getAddress(), response->getData(), response->getByteCount());
    } else {
      if (instance->_onError) instance->_onError(response->getError(),response->getSlaveAddress());
    }
    delete request;  // object created in public methods
    delete response;  // object created in _receive()
  }
}
```
**_In esp32ModbusTypeDefs.h:_**

`typedef std::function<void(esp32Modbus::Error,uint8_t)> MBRTUOnError;`

**_In the call:_**

```
modbus.onError([](esp32Modbus::Error error, uint8_t serverAddress) {
  });
```
_And no more modification._

### But i get address 0 always.

bertmelis, 2020-06-15:
You don't have to change the onError setter, you've already changed the handler's signature.

But the other changes seem to be correct. Don't know (yet) what is causing the slaveId to be always zero.

zekageri, 2020-06-17:
I don't know how should i get the address in the error handler. I really need it because i'am trying to handle multiple slaves and if one of them is not responding i have to delete it from an array wich is containing all the salves in the bus. I don't want to communicate with a dead slave. 

Did you investigated this issue? No matter what i do, can't get the slave address in the error handler. :|

bertmelis, 2020-06-17:
I'll look into the matter.

bertmelis, 2020-06-18:
So I've made a branch "develop" with the slaveAddress added to the onError callback.
Could you try that one?

zekageri, 2020-06-18:
Yes! Of course.

zekageri, 2020-06-18:
Still address 0. :/

```
static const inline void Handle_Error(String Code, String SlaveAddress){
  String Error = "Error Code: ";Error += Code;
  Error += " From Address: ";Error += SlaveAddress;
  Send_Async(Error," ~ModBus");
}
```

```
modbus.onError([](esp32Modbus::Error error, uint8_t slaveAddress) {
    Bus_is_Busy = false;
    if(Can_Check_Array){Check_Loop_Count();}
    Handle_Error(String(error,HEX),String(slaveAddress,HEX));
  });
```

closed 2020-06-23 by zekageri