bertmelis/esp32ModbusRTU #28: Sanity precaution

Miq1, 2020-08-08:
Hi Bert,

I may have found a potential reason for your ESP32MODBUSRTU task to crash after roughly 500 days on inactivity. This is unlikely to happen for a system used regularly, but anyway...
The xQueueReceive() call will ultimatively return after portMAX_DELAY ticks, which translates into the mentioned 500 days. The queue item in &request then is undefined. If one will regard the return code of xQueueReceive() -pdTRUE or pdFALSE -, the crash can be avoided. 

```
void esp32ModbusRTU::_handleConnection(esp32ModbusRTU* instance) {
  while (1) {
    ModbusRequest* request;
    if(pdTRUE==xQueueReceive(instance->_queue, &request, portMAX_DELAY))  // block and wait for queued item
    {
      instance->_send(request->getMessage(), request->getSize());
      ModbusResponse* response = instance->_receive(request);
      if (response->isSucces()) {
        if (instance->_onData) instance->_onData(response->getSlaveAddress(), response->getFunctionCode(), response->getData(), response->getByteCount(), response->getToken());
      } else {
        if (instance->_onError) instance->_onError(response->getError(), response->getToken());
      }
      delete request;  // object created in public methods
      delete response;  // object created in _receive()
    }
  }
}
```

bertmelis, 2020-08-10:
Yes! This also makes the code more understandable.

Miq1, 2020-08-11:
I started creating pull requests for the fixes and amendments I have made. Please bear with me as I am new to git and especially seem to have issues with the cpplint rules Travis CI is applying. That led to the first small pull request containing 3 failed and one successful commit - all for lint reasons... <*blush*>

bertmelis, 2020-08-11:
You've pushed all your changed into one branch I think. So all your commits end up in both PRs.

I've got the one from this issue merged.

The other I'm not sure of regarding the pin. Let's discuss in the PR itself.

closed 2020-08-11 by bertmelis