bertmelis/esp32ModbusRTU #40: New receive function issue

Miq1, 2020-08-20:
Hi Bert,

I need your advice, please... I finished implementing the new, bus timing-based receive function, but it shows some odd behaviour with larger responses. Everything that fits into 112 bytes runs fine, if the response is longer, receive will fail.

The essential part of the ``receive()`` function are the two states shown below. ``BUFBLOCKSIZE`` is 128 and ``buffer`` is initialized with one block of 128 bytes initially. ``bufferPtr`` starts at 0:
```
    // IN_PACKET: read data until a gap of at least _interval time passed without another byte arriving
    case IN_PACKET:
      // Data waiting and space left in buffer?
      while (_serial->available()) {
        // Yes. Catch the byte
        buffer[bufferPtr++] = _serial->read();
        // Buffer full?
        if(bufferPtr >= bufferBlocks * BUFBLOCKSIZE) {
          // Yes. Extend it by another block
          bufferBlocks++;
          uint8_t *temp = new uint8_t[bufferBlocks * BUFBLOCKSIZE];
          memcpy(temp, buffer, (bufferBlocks - 1) * BUFBLOCKSIZE);
          delete buffer;
          buffer = temp;
        }
        // Rewind timer
        _lastMicros = micros();
      }
      delayMicroseconds(1);
      // Gap of at least _interval micro seconds passed without data?
      if(micros() - _lastMicros >= _interval) {
        state = DATA_READ;
      }
      break;
    // DATA_READ: successfully gathered some data. Prepare return object.
    case DATA_READ:
      // Allocate response object
      response = new ModbusResponse(bufferPtr, request);
      // Move gathered data into it
      response->setData(bufferPtr, buffer);
      state = FINISHED;
      break;
```
The data collected in buffer is moved into the response's ``_buffer`` with a call to 
```
void ModbusResponse::setData(uint16_t dataLength, uint8_t *data) {
  if(_length != dataLength) {
    delete _buffer;
    _buffer = new uint8_t[dataLength];
    _length = dataLength;
  }
  _index = dataLength;
  memcpy(_buffer, data, dataLength);
}
```
The effect observable with long responses is illustrated with this test output caught in the checkCRC function:#
```
18:15:05.356 CRC ERR: _length=112 CRC calc=A92B CRC pack=00 ?
18:15:05.367 04 03 80 01 84 C4 5A C0 00 47 1D 53 CD 49 05 FE
18:15:05.380 1A 47 1D 53 CD 49 05 FE 1A 00 00 00 00 00 00 00
18:15:05.393 00 45 4D 48 00 09 01 45 4D 48 00 00 90 2F 2C 00
18:15:05.406 00 00 00 00 06 9C 68 00 00 09 6F 00 00 00 00 00
18:15:05.418 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
18:15:05.431 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
18:15:05.445 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00 00
```
The packets are from slave #04, function code is 0x03 and the data length to be expected is 0x80, as the first three bytes show. Subsequently the bytes up to the 112th - the ``_buffer`` size ``_length`` is indicating - are correctly shown.

But why does it stop at 112? To be expected are 128 (data) + 3 (slave, function code and length) + 2 (CRC) = 135. Clearly this will not fit in the initial buffer size of 128 that needs to be extended. But this will happen only if ``bufferPtr`` gets to 128, which is still 16 bytes ahead.

``_lastMicros`` will be set to ``micros()`` immediately before the while loop breaks, so the timeout should not have struck prematurely. The only way to fall into that hole would be ``_serial->available()`` lasting more than ``_interval`` microseconds - unlikely, I would say. I nevertheless tried with an ``_interval`` of 4000 µs but the effect persisted.

I am at a loss here, I am afraid.

Miq1, 2020-08-20:
I might have found a hint on the 112 here: [https://github.com/espressif/arduino-esp32/pull/3664](url)

It looks like the FIFO buffer of HardwareSerial has a high water mark at 112 bytes. This would indicate I am causing a FIFO overflow somehow. But why? I do not think the while loop could be too slow with these few instructions in it. I am running at 19200 baud, so plenty of time, 

Another reason could be preventing the UART ISR to happen, but again: what should cause this? 😕

Miq1, 2020-08-24:
Closing this as the workaround is functional and it rather seems to be an issue in the arduino-esp32 core. Issue report created there as [https://github.com/espressif/arduino-esp32/issues/4281](url).

closed 2020-08-24 by Miq1

bertmelis, 2020-08-24:
I'll do some more basic testing. But time is very limited ATM.

Miq1, 2020-08-24:
For whatever it is worth, the code now is running flawlessly on my bridge device for a couple of days. I am only doing FC 02, 03 and a custom one, though, so I do not know if the writing FCs have survived the changes. I do not see why not, the relevant code was untouched and the remainder is used by the other FCs regardless.