bertmelis/esp32ModbusRTU #45: custom modbus request

EmanueleFeola, 2020-10-01:
How can I implement with the library a request like this?
[image]

bertmelis, 2020-10-02:
This is not a modbus message format.

While it should not be a big problem implementing this in this lib, it is very specific and out of scope.

EmanueleFeola, 2020-10-02:
For now, I came up with this, but I can't understand what to assign to _byteCount
Can you please give me a hint?

[image]

bertmelis, 2020-10-02:
`_byteCount` is used to calculate the length of the response message (and add the CRC into the right location).

I don't know what type of response you get when using this message.

EmanueleFeola, 2020-10-02:
This is the response I get.
So should _byteCount be 6? 

[image]

EmanueleFeola, 2020-10-05:
I wrote the function without using the library, problem solved

closed 2020-10-05 by EmanueleFeola