bertmelis/esp32ModbusRTU #46: error: 0xe3

Tsaadon, 2020-10-04:
Hey,
I'm trying to connect a wind speed sensor using the code provided, I changed the address to match the sensor factory default but getting error: 0xe3, what does it mean and how can I resolve it?
By the way, I've tested the sensor using USB and modbus software and it works well.
[Wind speed 485 transmitter instructions.pdf](https://github.com/bertmelis/esp32ModbusRTU/files/5322906/Wind.speed.485.transmitter.instructions.pdf)

`#include <Arduino.h>
#include <esp32ModbusRTU.h>
#include <algorithm>  // for std::reverse

esp32ModbusRTU modbus(&Serial1, 15);  // use Serial1 and pin 16 as RTS

void setup() {
  Serial.begin(4800);  // Serial output
  Serial1.begin(4800, SERIAL_8N1, 4, 2, true);  // Modbus connection

  modbus.onData([](uint8_t serverAddress, esp32Modbus::FunctionCode fc, uint16_t address, uint8_t* data, size_t length) {
    Serial.printf("id 0x%02x fc 0x%02x len %u: 0x", serverAddress, fc, length);
    for (size_t i = 0; i < length; ++i) {
      Serial.printf("%02x", data[i]);
    }
    std::reverse(data, data + 4);  // fix endianness
    Serial.printf("\nval: %.2f", *reinterpret_cast<float*>(data));
    Serial.print("\n\n");
  });
  modbus.onError([](esp32Modbus::Error error) {
    Serial.printf("error: 0x%02x\n\n", static_cast<uint8_t>(error));
  });
  modbus.begin();

}

void loop() {
  static uint32_t lastMillis = 0;
  if (millis() - lastMillis > 5000) {
    lastMillis = millis();
    Serial.print("sending Modbus request...\n");
    modbus.readInputRegisters(0x01, 0, 1);
  }
}`

Thanks!

Hasenburg, 2020-10-05:
0xE3 is a CRC Error.
Both of your Serials run on baud rate 4800.

Change Serial.begin(4800) to Serial.begin(9600)

and try 

Serial1.begin(4800, SERIAL_8N1, 4, 2, **false**);

Tsaadon, 2020-10-05:
Ok, did as you said and now I get this:

18:13:57.062 -> ⸮⸮⸮⤤⸮⸮⸮⸮⸮g⸮⸮儦esending Modbus request...
18:14:02.112 -> error: 0xe3
18:14:02.145 -> 
18:14:07.066 -> sending Modbus request...
18:14:07.307 -> error: 0x57
18:14:07.307 -> 
18:14:12.077 -> sending Modbus request...
18:14:12.247 -> error: 0xe3
18:14:12.281 -> 
18:14:17.055 -> sending Modbus request...
18:14:17.191 -> error: 0xe3
18:14:17.191 -> 
18:14:22.081 -> sending Modbus request...
18:14:22.149 -> id 0x00 fc 0xfe len 0: 0x
18:14:22.183 -> val: -2678576141896654202868169793836417024.00
18:14:22.217 -> 
18:14:27.056 -> sending Modbus request...
18:14:27.156 -> id 0x00 fc 0xff len 0: 0x
18:14:27.192 -> val: -2679062919727141842958344527867281408.00
18:14:27.225 -> 
18:14:32.086 -> sending Modbus request...
18:14:32.154 -> id 0x00 fc 0xff len 0: 0x
18:14:32.188 -> val: -10716576197462225798560161267489701888.00
18:14:32.223 -> 
18:14:37.063 -> sending Modbus request...
18:14:37.131 -> id 0x00 fc 0xff len 0: 0x
18:14:37.165 -> val: -42535295865117307932921825928971026432.00
18:14:37.232 -> 
18:14:42.085 -> sending Modbus request...
18:14:42.154 -> id 0x00 fc 0xff len 0: 0x
18:14:42.188 -> val: -171449642268820008294076988790847569920.00
18:14:42.256 -> 
18:14:47.058 -> sending Modbus request...
18:14:47.159 -> error: 0xe3
18:14:47.159 -> 
18:14:52.073 -> sending Modbus request...
18:14:52.140 -> id 0x00 fc 0xff len 0: 0x
18:14:52.174 -> val: 0.00
18:14:52.174 -> 
18:14:57.058 -> sending Modbus request...
18:14:57.161 -> id 0x00 fc 0xff len 0: 0x
18:14:57.194 -> val: -42865006715634269487333512445876502528.00
18:14:57.228 -> 
18:15:02.073 -> sending Modbus request...
18:15:02.174 -> id 0x00 fc 0xff len 0: 0x
18:15:02.208 -> val: -10465406442834204881406407487782912.00
18:15:02.241 -> 
18:15:07.088 -> sending Modbus request...
18:15:07.156 -> id 0x00 fc 0xff len 0: 0x
18:15:07.191 -> val: -171460026862537077949334049783506010112.00
18:15:07.259 -> 
18:15:12.072 -> sending Modbus request...
18:15:12.173 -> id 0x00 fc 0xfe len 0: 0x
18:15:12.207 -> val: -42866304789848903194240645069958807552.00
18:15:12.240 -> 
18:15:17.084 -> sending Modbus request...
18:15:17.185 -> id 0x00 fc 0xfd len 0: 0x
18:15:17.218 -> val: -42865006715634269487333512445876502528.00
18:15:17.252 -> 
18:15:22.069 -> sending Modbus request...
18:15:22.172 -> id 0x00 fc 0xff len 0: 0x
18:15:22.207 -> val: -42862410567205002073519247197711892480.00
18:15:22.241 -> 
18:15:27.090 -> sending Modbus request...
18:15:27.159 -> error: 0xe3
18:15:27.159 -> 
18:15:32.081 -> sending Modbus request...
18:15:32.183 -> id 0x00 fc 0xfd len 0: 0x
18:15:32.183 -> val: -10716576197462225798560161267489701888.00
18:15:32.252 -> 
18:15:37.086 -> sending Modbus request...
18:15:37.188 -> id 0x00 fc 0xff len 0: 0x
18:15:37.188 -> val: -171460026862537077949334049783506010112.00
18:15:37.257 -> 
18:15:42.098 -> sending Modbus request...
18:15:42.166 -> id 0x00 fc 0xff len 0: 0x
18:15:42.201 -> val: -167446503085347278102502519804526592.00
18:15:42.236 -> 
18:15:47.090 -> sending Modbus request...
18:15:47.191 -> id 0x00 fc 0xfe len 0: 0x
18:15:47.225 -> val: -171449642268820008294076988790847569920.00
18:15:47.259 ->

Hasenburg, 2020-10-05:
You are using readInputRegisters which has function code 0x02. 
You Wind Speed thingy accepts 0x03, which is
modbus.readHoldingRegisters(slaveaddress, address, data)

Tsaadon, 2020-10-05:
Ok, changed it, still get the same issue:

18:32:30.692 -> sending Modbus request...
18:32:30.759 -> error: 0xe3
18:32:30.793 -> 
18:32:35.672 -> sending Modbus request...
18:32:35.773 -> id 0x00 fc 0xff len 0: 0x
18:32:35.807 -> val: -2679062919727141842958344527867281408.00
18:32:35.842 -> 
18:32:40.667 -> sending Modbus request...
18:32:40.768 -> id 0x00 fc 0xfa len 0: 0x
18:32:40.802 -> val: -42866304789848903194240645069958807552.00
18:32:40.871 -> 
18:32:45.674 -> sending Modbus request...
18:32:45.776 -> id 0x00 fc 0xfe len 0: 0x
18:32:45.810 -> val: -2679144049365556449640040316872425472.00
18:32:45.845 -> 
18:32:50.698 -> sending Modbus request...
18:32:50.767 -> error: 0xe3
18:32:50.767 -> 
18:32:55.688 -> sending Modbus request...
18:32:55.755 -> id 0x00 fc 0xfe len 0: 0x
18:32:55.789 -> val: -10715927160354908945106594955448549376.00
18:32:55.857 -> 
18:33:00.688 -> sending Modbus request...
18:33:00.790 -> id 0x00 fc 0xff len 0: 0x
18:33:00.790 -> val: -2679062919727141842958344527867281408.00
18:33:00.857 ->

bertmelis, 2020-10-11:
Try this, I took it straight from your manual.
If it doesn't work, also try swapping the A and B wires.

The sample code assumes returned data is 4 bytes and you only get 2 bytes.

```
#include <Arduino.h>
#include <esp32ModbusRTU.h>

esp32ModbusRTU modbus(&Serial1, 15);  // use Serial1 and pin 16 as RTS

void setup() {
  Serial.begin(4800);  // Serial output
  Serial1.begin(4800, SERIAL_8N1, 4, 2, true);  // <-- change this to your settings!!!!

  modbus.onData([](uint8_t serverAddress, esp32Modbus::FunctionCode fc, uint16_t address, uint8_t* data, size_t length) {
    Serial.printf("id 0x%02x fc 0x%02x len %u: 0x", serverAddress, fc, length);
    for (size_t i = 0; i < length; ++i) {
      Serial.printf("%02x", data[i]);
    }
    uint16_t temp = 0;
    memcpy(&temp, data, 2);
    float windSpeed = temp / 10.0;
    Serial.printf("\nwind speed: %.2f", windSpeed);
    Serial.print("\n\n");
  });
  modbus.onError([](esp32Modbus::Error error) {
    Serial.printf("error: 0x%02x\n\n", static_cast<uint8_t>(error));
  });
  modbus.begin();

}

void loop() {
  static uint32_t lastMillis = 0;
  if (millis() - lastMillis > 5000) {
    lastMillis = millis();
    Serial.print("sending Modbus request...\n");
    modbus.readHoldingRegisters(0x01, 0, 1);  // FC03, id 1, address 0, 1 register
  }
}
```

Tsaadon, 2020-10-12:
Ok so when trying with the code you provided this is what I got:
`13:56:13.452 -> sending Modbus request...
13:56:13.554 -> error: 0xe3
13:56:13.588 -> 
13:56:18.432 -> sending Modbus request...
13:56:18.534 -> error: 0xe3
13:56:18.568 -> 
13:56:23.430 -> sending Modbus request...
13:56:23.498 -> error: 0xe3
13:56:23.531 -> 
13:56:28.451 -> sending Modbus request...
13:56:28.552 -> error: 0xe3
13:56:28.586 -> 
13:56:33.460 -> sending Modbus request...
13:56:33.563 -> error: 0x32
13:56:33.563 -> 
13:56:38.451 -> sending Modbus request...
13:56:38.553 -> error: 0xe3
13:56:38.587 -> 
13:56:43.446 -> sending Modbus request...
13:56:43.547 -> error: 0xe3
13:56:43.580 -> 
13:56:48.435 -> sending Modbus request...
13:56:48.502 -> error: 0x01
13:56:48.535 -> 
13:56:53.448 -> sending Modbus request...
13:56:53.550 -> error: 0x5a
13:56:53.583 -> 
13:56:58.447 -> sending Modbus request...
13:56:58.516 -> error: 0xe3
13:56:58.516 -> 
13:57:03.451 -> sending Modbus request...
13:57:03.520 -> error: 0x29
13:57:03.520 -> 
13:57:08.437 -> sending Modbus request...
13:57:08.541 -> error: 0xe3
13:57:08.575 -> 
13:57:13.459 -> sending Modbus request...
13:57:13.561 -> error: 0xe3
13:57:13.596 -> 
13:57:18.450 -> sending Modbus request...
13:57:18.555 -> error: 0x26
13:57:18.589 -> 
13:57:23.440 -> sending Modbus request...
13:57:23.575 -> error: 0x7a
13:57:23.575 -> 
13:57:28.449 -> sending Modbus request...
13:57:28.552 -> error: 0xe3
13:57:28.585 -> 
13:57:33.446 -> sending Modbus request...
13:57:33.548 -> error: 0xe3
13:57:33.582 -> 
13:57:38.445 -> sending Modbus request...
13:57:38.547 -> error: 0xe3
13:57:38.580 -> 
13:57:43.462 -> sending Modbus request...
13:57:43.496 -> error: 0xe3
13:57:43.529 -> 
13:57:48.456 -> sending Modbus request...
13:57:48.556 -> error: 0xe3
13:57:48.589 -> 
13:57:53.464 -> sending Modbus request...
13:57:53.565 -> error: 0xe3
13:57:53.599 -> 
13:57:58.473 -> sending Modbus request...
13:57:58.574 -> error: 0xe3
13:57:58.574 -> 
13:58:03.450 -> sending Modbus request...
13:58:03.551 -> error: 0xe3
13:58:03.586 -> 
13:58:08.471 -> sending Modbus request...
13:58:08.505 -> error: 0x01
13:58:08.539 -> 
13:58:13.445 -> sending Modbus request...
13:58:13.580 -> id 0x00 fc 0x95 len 0: 0x
13:58:13.614 -> wind speed: 4456.00
13:58:13.648 -> 
13:58:18.449 -> sending Modbus request...
13:58:18.552 -> error: 0xe3
13:58:18.586 -> 
13:58:23.458 -> sending Modbus request...
13:58:23.560 -> error: 0xe3
13:58:23.594 -> 
13:58:28.460 -> sending Modbus request...
13:58:28.563 -> error: 0xe3
13:58:28.597 -> 
13:58:33.447 -> sending Modbus request...
13:58:33.583 -> error: 0xe3
13:58:33.583 -> 
13:58:38.455 -> sending Modbus request...
13:58:38.559 -> error: 0xe3
13:58:38.594 -> 
13:58:43.465 -> sending Modbus request...
13:58:43.567 -> error: 0xe3
13:58:43.601 -> 

`

I also tried with changing the  "true" to "false" and this is what I get, only the "error: 0xe3":
`13:59:31.091 -> sending Modbus request...
13:59:31.261 -> error: 0xe3
13:59:31.261 -> 
13:59:36.123 -> sending Modbus request...
13:59:36.262 -> error: 0xe3
13:59:36.297 -> 
13:59:41.111 -> sending Modbus request...
13:59:41.282 -> error: 0xe3
13:59:41.282 -> 
13:59:46.125 -> sending Modbus request...
13:59:46.260 -> error: 0xe3
13:59:46.294 -> 
13:59:51.116 -> sending Modbus request...
13:59:51.253 -> error: 0xe3
13:59:51.287 -> 
13:59:56.114 -> sending Modbus request...
13:59:56.282 -> error: 0xe3
13:59:56.282 -> 
14:00:01.109 -> sending Modbus request...
14:00:01.279 -> error: 0xe3
14:00:01.279 -> 
14:00:06.099 -> sending Modbus request...
14:00:06.268 -> error: 0xe3
14:00:06.302 -> 
14:00:11.118 -> sending Modbus request...
14:00:11.255 -> error: 0xe3
14:00:11.255 -> 
14:00:16.098 -> sending Modbus request...
14:00:16.234 -> error: 0xe3
14:00:16.269 -> 
14:00:21.115 -> sending Modbus request...
14:00:21.252 -> error: 0xe3
14:00:21.286 -> 
`

Tried swapping A and B, didn't work either.. any other ideas?

bertmelis, 2020-10-12:
What components do you use? Does your level converter support 3.3V? How does your circuit looks like?

Tsaadon, 2020-10-12:
Sparkfun RS485 Breakout:
https://www.sparkfun.com/products/10124

ESP32S:
https://cdn.instructables.com/ORIG/FOL/YWLI/JEOILQ5U/FOLYWLIJEOILQ5U.png?fit=bounds&frame=1&width=1024

I connected an external 12V power supply for the sensor, the breakout I'm feeding 3.3v with the esp32. I attached a zip with pics of the setup.

I highly appreciate the help!

[Pics.zip](https://github.com/bertmelis/esp32ModbusRTU/files/5365319/Pics.zip)

bertmelis, 2020-10-12:
Is your Arduino framework up-to-date? Are you using Arduino IDE or Platformio?

Regarding hardware: did you add the 120Ohms terminating resistor at the device side?
I'm not sure if the Sparkfun board has it or not.

Hasenburg, 2020-10-12:
I ve noticed two things with your setup:

#1 In your code you use the following Pins:
RTS = Pin 15
Rx = Pin 4
Tx = Pin 2

In the photo you wired your Hardware as followed: 
RTS = Pin 34
Rx = Pin 32
Tx = Pin 35

Those should match.

#2 You Sensor is not Grounded to the ESP32.

Tsaadon, 2020-10-13:
bertmelis:
No actually I do get a message I need to update my Arduino IDE, however, I'm afraid of doing so as I have much in stake now and  I don't want a software update to mess up with all my other sensor..

Regarding the resistor, I didn't use any and I'm not sure if the Sparkfun has one either, but I can say that for another RS485 sensor everything is working fine with the Sparkfun breakout board.

Hasenburg:
Yes I changed the pins and updated everything, also tried other pins and no success either..

It's grounded using the 12V adapter, you can see in the picture.

I'm at loss here.. if you have any more ideas I'd love to hear, in any case, I highly appreciate both of your help!

Hasenburg, 2020-10-13:
Ok, but I guess is the 12V adapter is not grounded to the ESP32? 
The ESP32 and the Sensor, need to have the same GND!!
You need to add a wire between the Sensor's GND (or 12V adapter GND) and the 'G' Pin of the RS485 board(next to A Pin). 
If they don't have the same GND, they have different logic levels hence they don't acknowledge them as message, which leads to timeout errors.

Tsaadon, 2020-10-13:
Ohhh... I'm an electronics newbie :) now it works... I still get "error: 0xe3" every now and then, but it's good enough, most of the time its good.. thanks!!

The values I get are weird though, much much higher (about 100 times) than they should be, I checked the hex output and it doesn't correlate with the decimal output, the hex values seem ok but something is going bad in the transition to decimal, any idea?

This is the output:
16:22:56.923 -> sending Modbus request...
16:22:56.958 -> id 0x01 fc 0x03 len 2: 0x0010
16:22:56.993 -> wind speed: 409.60

From what I see 0x0010 is 16 and not 409.

This is the code that is working:

`#include <Arduino.h>
#include <esp32ModbusRTU.h>

#define RS485_RX_PIN 14
#define RS485_TX_PIN 13
#define RS485_DERE_PIN 15

esp32ModbusRTU modbus(&Serial1, RS485_DERE_PIN);  // use Serial1 and pin 34 as RTS

void setup() {
  Serial.begin(9600);  // Serial output
  Serial1.begin(4800, SERIAL_8N1, RS485_RX_PIN, RS485_TX_PIN, false);  // <-- change this to your settings!!!!

  modbus.onData([](uint8_t serverAddress, esp32Modbus::FunctionCode fc, uint16_t address, uint8_t* data, size_t length) {
    Serial.printf("id 0x%02x fc 0x%02x len %u: 0x", serverAddress, fc, length);
    for (size_t i = 0; i < length; ++i) {
      Serial.printf("%02x", data[i]);
    }
    uint16_t temp = 0;
    memcpy(&temp, data, 2);
    float windSpeed = temp / 10.0;
    Serial.printf("\nwind speed: %.2f", windSpeed);
    Serial.print("\n\n");
  });
  modbus.onError([](esp32Modbus::Error error) {
    Serial.printf("error: 0x%02x\n\n", static_cast<uint8_t>(error));
  });
  modbus.begin();

}

void loop() {
  static uint32_t lastMillis = 0;
  if (millis() - lastMillis > 5000) {
    lastMillis = millis();
    Serial.print("sending Modbus request...\n");
    modbus.readHoldingRegisters(0x01, 0, 1);  // FC03, id 1, address 0, 1 register
  }
}`

Tsaadon, 2020-10-13:
Ok a friend helped me out, we changed:
memcpy(&temp, data, 2);

To:
uint16_t temp = (data[0] << 8) | data[1];

and now it's working perfect, thanks! you guys are they best.

bertmelis, 2020-10-13:
Regarding the terminating resistor: the manual you attached does mention this but it refers to another document. It depends on whether you have more then one modbus device on your wire.

Side note: if you're connecting grounds, you may want to do it via a resistor to limit the current if there are any large differences.

closed 2020-10-14 by bertmelis