bertmelis/esp32ModbusRTU #47: error: 0xe3

EmanueleFeola, 2020-10-19:
Hi, I sometimes receive this error:
error: 0xe3 slaveId: 8 fc 0x10 address 0x27d

From the source code of the library I understand that it is a CRC problem, but I don't understand why it gets thrown.
The function that generates the error is the WRITE_MULT_REGISTERS.

But if the problem is related to the CRC, shouldn't the problem appear in all write requests?
I mean, I don't think the error is thrown because the code calculates the wrong CRC

bertmelis, 2020-10-19:
Everything else is working fine?
Are you also using other function codes?
Maybe it's hardware related? (level converter, line termination, line driver...)

EmanueleFeola, 2020-10-19:
Sometimes I receive these errors too...but very rarely.
error: 0x0b slaveId: 3 fc 0x90 address 0x103a
error: 0x10 slaveId: 8 fc 0xff address 0x27d
"Are you also using other function codes?" yes, I'm using the following functions:
readHoldingRegisters
writeSingleHoldingRegister
writeMultHoldingRegisters

"Maybe it's hardware related?" All I have of hw is a rx-tx-gnd-vcc cable connected to a electric board (connected to an electric engine)

bertmelis, 2020-10-19:
> "Maybe it's hardware related?" All I have of hw is a rx-tx-gnd-vcc cable connected to a electric board (connected to an electric engine)

You mean you connected the esp directly to a modbus line? Normal way is to use a line driver like shown on the readme page.
Are you sure that you receive correct values when there's no error?

EmanueleFeola, 2020-10-19:
Yes, I'm sure that I receive correct values, and the bytes I send are received correctly
This is a photo of the cable
[image]

EmanueleFeola, 2020-11-06:
Solved (not related to modbus communication)

closed 2020-11-06 by EmanueleFeola